moschotto/OpenTX_GPS_Telemetry #1: Does not work on Radiomaster TX16s or Horus etc...

devosthomas, 2020-12-21:
I am using your script for several years on my X9D Taranis but I recently switched, like a lot of people, to a Radiomaster Tx16s.

Unfortunately when I execute the script in Tools it throws an error:

"bad argument #6 to 'drawLine' (number expected, got nil)"

I guess it is because the Radiomaster/Horus and others don't doesn't support some flags that work with Taranis. The following are not defined on the Radiomaster/Horus but used commonly with the Taranis:

GREY_DEFAULT
FORCE
ERASE

See also post: https://www.rcgroups.com/forums/showthread.php?2988482-making-the-Horus-Graph-script-usable-for-the-Taranis

Could you change it? I have both radio's so I can test if you want.

moschotto, 2020-12-23:
it's not just the draw line....  check the repository again -> T16

closed 2020-12-23 by moschotto

devosthomas, 2020-12-26:
Wow great! Just tested and works like a charm!

Maybe an idea which would be great. Create a QR code with the coordinates. You scan the code using your smartphone and Google maps app shows your location. QR code contains dynamic url e.g. : https://www.google.be/maps/@51.1665956,3.328285

No more typing in coordinates. 

Seems you can generate QR's using Lua https://github.com/speedata/luaqrcode