moschotto/OpenTX_GPS_Telemetry #2: Script Syntax Error

hgg2k, 2020-12-28:
I am using the X9D Radio.  First of all I had to copy the lua script in the TELEMETRY folder and not
in the TOOLS subfolder because in the setup of the 4 telemetry screens it only show the scripts that 
are in the TELEMETRY folder.

When I switch to that screen and the lua script gets executed I get the following error:

**Script syntax error
TELEMETRY/GPSvX9.lua:61
bad argument #1 to 'gsub' (string expec**

I have also renamed the script to GPSvX9.lua because it does not accept more than 8 characters.

Thanks!

moschotto, 2020-12-28:
please read the instructions again:

no need to rename anything - you probably copied the wrong files!

Install Taranis models:

    copy GPSxxx.lua to /SCRIPTS/TELEMETRY

    copy GPS**viewer**xxx.lua to /SCRIPTS/TOOLS

The GPS Viewer Script goes into the TOOL folder and not into the TELEMETRY folder. 

BTW. during uploading i move some files in the wrong folder. corrected this already...

closed 2020-12-28 by moschotto

hgg2k, 2020-12-28:
Hello,

I do not have a TOOLS subfolder inside the SCRIPTS folder in my radio and when I create
one and place the GPS viewer inside then in my telemetry screen I cannot add the script.
It only shows the scripts that are in the TELEMETRY folder.

Also if I do not rename the file GPSviewerX9.lua to something like GPSvX9.lua it will not
work, because my radio works only with 8 character filenames...

Did you change the format of the GPSpositions.txt file in the new version?

Thank you.
George.

hgg2k, 2020-12-28:
Ok,

I have deleted all previous versions and files,  I have copied everything as per instructions
and now I get the following error:

**Script syntax error
TELEMETRY/GPSx9.lua:84
attempt to call global 'getGlobalTimer'**

moschotto, 2020-12-28:
**Regarding the script error:**

getGlobalTimer()

Returns radio timers

@status current Introduced added in 2.3.0.

so please make sure that your radio is a least on OpenTX version 2.3!

**Regarding the TOOLS folder:**

if you download SDCARDS from here:
https://downloads.open-tx.org/2.3/release/sdcard/

you can see that the TOOLS folder exists under SCRIPTS

hgg2k, 2020-12-28:
My radio has OpenTX version 2.2
So there is no way to run the script on this version?  :-(

moschotto, 2020-12-28:
Should be no problem to update the radio via the companion app

hgg2k, 2020-12-28:
I am afraid to lose my models setup...
Anyway, I will try to update.

Great work on the GPS logging!   Life Saver!
Thanks.

moschotto, 2020-12-28:
Thank and BTW you can backup your models with companion app

hgg2k, 2020-12-28:
I set **time_power_on=0** and copied the viewer script in the TELEMETRY folder with an 8 character
filename.  So, now it works in OpenTX v2.2 as well but without displaying the time.  

When I find the courage, I might upgrade to 2.3 as well.  :-)
But for now, all good!    Time to test the new HD LR build.   :thumbsup:

moschotto, 2020-12-28:
sorry i don't get - what's the problem???

the telemetry script (NO RENAMING necessary):
/SCRIPTS/TELEMETRY/GPSx9.lua

the logile viewer:
/SCRIPTS/TOOLS/GPSviewerX9.lua 

the VIEWER is just to check/view the logfile which is generated by the telemetry script. check the "Demo Video" again.

hgg2k, 2020-12-28:
There is no problem now.
Maybe you do not have the 2.2 version.

**"Due to changes introduced by Horus, images, sounds, and scripts files names for ALL radio are restricted to 6 characters in most places."  http://www.open-tx.org/2019/01/06/opentx-2.2.3**
 
So I had to give a filename with fewer number of characters like **GPSvX9.lua** which can be recognized
by OpenTX 2.2   

Both the telemetry script and the viewer are working fine now in OpenTX 2.2 with the exception
of not displaying the session timer.  I changed the script to the following:

	--time_power_on = SecondsToClock(getGlobalTimer()["session"])
						
		--write logfile		
		file = io.open(log_filename, "a")    						
		io.write(file, coordinates_current ..",".. 0,"\r\n")	

Its a compromise but it works now.