moschotto/OpenTX_GPS_Telemetry #3: Just a question about the LUA Script

zodi1337, 2021-01-08:
Hi moschotto

I'm interested in using your script for my GPS-Quad with my Taranis X7S.

Since i'm relatively new in this hobby and don't get much experience with telemetry and LUA scripts here my question:

Right now i'm using OpenTx interrnal SD-Log procedure. This just write a logfile from my telemetry sensors to the SDcard at my Taranis. In my case it sends writes all telemetry data every 1 second. I use this to reproduce my flight routes and also to see the actual GPS-coordinates on the Taranis.

I'm just a bit unsure, how you get the GPS data and write it to your own logfile and how often?
Do you also capture the GPS data and send it via telemetry?

Since i'm get already my GPS data every 1 second in my OpenTx logfile (csv), this would mean if i use your script the telemetry data is sent twice. One time for the OpenTx log and one time for your LUA-Script.

If this is the case we could maybe think about a procedure to use the standard OpenTx SDLog function and let your LUA script just read data from the OpenTx csv logfile?

Sorry if i get something totally wrong, these are just my 2 cents about the script.

Looking forward to test it!

moschotto, 2021-01-08:
"I'm just a bit unsure, how you get the GPS data and write it to your own logfile and how often?
Do you also capture the GPS data and send it via telemetry?"

-> The telemetry is transmitted FROM the quad (flight controller) to the radio continuously,- it won't be "pulled". all data readings will be performed "within the radio", so it doesn't matter how many scripts access the data.  

OpenTx internal SD-Log procedures log all data to a separate lofile. My script "OpenTX_GPS_Telemetry" stores only GPS coordinates and radio-power-on time into the a separate logfile. If you want to have full flight logs (incl all available sensors), then use the build in logging function.

--------------------------

"If this is the case we could maybe think about a procedure to use the standard OpenTx SDLog function and let your LUA script just read data from the OpenTx csv logfile?"

No, because not everyone switches on telemetry logging (like myself).

--------------------------

Further readings:
https://oscarliang.com/log-gps-coordinates-taranis/

If you don’t have access to a computer in the field, most smart phones with micro SD card slot can read it as well.
You can also access the log in OpenTX Companion.

-> As mentioned in the description: the intention is to have a "onboard" **live** GPS telem screen and a GPS logfile viewer on the radio,

closed 2021-01-08 by moschotto

zonArt, 2021-05-26:
Hi,

Jumping on this issue which is the most linked to my actual question (if it's a bad idea, feel free to tell me to open a new issue). I'm also using opentx telemetry logging capability and took your viewer script which I adapted to read the telemetry log file instead and display other informations such as LQ, RSSI, speed, RX power and quad battery. Well this works quite good but I have two questions: 
- do you know if it is possible to list files in a directory to be able to chose which file you want to read (for now I've hardcoded a filename)?
- do you think you can read a file entirely? I mean, the io.read() function wants a length and it only reads from the beginning and the length is limited in the fact that the transmitter has a limited amount of memory. How could we read a chunk of data the move to a new one and so on until there is no more data?

Thank you very much for your work and your answers