moschotto/OpenTX_GPS_Telemetry #6: Text Color Change P{ossible?

TMac54, 2021-01-16:
This is not an issue.  I was just wondering if there's a way I could change the color of the text displayed on the screen by going into the code and changing a value(s).

The black color text works great on the standard background screen, however, with a dark colored custom background screen, the black text is almost unreadable.  Was hoping I could change it to white

Is there a rather simple way of doing this in the code or is it more complex?
Thank you!

moschotto, 2021-01-17:
done, new main.lua in the T16 folder. Checkout 2.2 release

closed 2021-01-17 by moschotto

TMac54, 2021-01-17:
> done, new main.lua in the T16 folder. Checkout 2.2 release

Thank you.  I see where you have added:
local options = {
  { "LineColor", COLOR, WHITE },
  { "TextColor", COLOR, BLACK }
in the new 2.2. release.  I have copied this new release onto SD card and folder per instructions as it is with "BLACK" . I have also changed BLACK to WHITE in this line of code.  Neither one of these changes the text color on my screen (it stays black).  Perhaps I am missing something.  
I appreciate your efforts and am not in any way complaining.  I can always just change my background image if need be to accommodate black text.  Thanks again!

moschotto, 2021-01-17:
Widget options -> select the color you want.

TMac54, 2021-01-17:
> When you setup the widget, press enter and select the color you want.

Perfect! Thank you! Works perfectly!  After I replaced the "old Main" with your new 2.2 main I did not setup the widget again (I was just using the old widget I had), my bad.  Thank you so much!  👍