moschotto/OpenTX_GPS_Telemetry #8: Widget Triggers emergency mode on setup

j-watkiss, 2021-01-23:
Noticed this today setting up. Specifically in the three view option on a TX16S running 2.3.11 with a 32GB Sandisk card.
Widget works fine when selected for the correct screen view (the larger right hand side) however, setting up other widgets on the two smaller views, scrolling past this widget in the bottom left view triggers emergency mode reliably. Does not prevent use or function provided you don't do this that I have noticed

moschotto, 2021-01-23:
I don’t own a T16 an therefore I probably won’t change the widget. Maybe if I find time  in the future. As mentioned in the description, the Widget should be used 1/2 screen - it does not scale up or down automatically.

j-watkiss, 2021-01-23:
No worries - if I had the capability I would look into it myself. No major issue, just a little quirk. Ok so long as you dont need a widget which is after this in the queue.

Thanks for your work on this, regardless. A much simpler solution than any we had before

closed 2021-01-23 by moschotto

moschotto, 2021-01-30:
hi, can you please checkout the newer main.lua in the T16 folder and let me know if this works for you? Also if the script crashes after setting up and then power off/on the radio?

The simulator seems to work fine

[image]

Thanks

roligmue, 2021-02-01:
Per your request, I loaded the updated script into my Horus X10. It works fine. I did not recreate the widget.                                                            

I will edit per your instructions to change km to miles for my needs. 

Regards

moschotto, 2021-02-01:
Hi thanks for the reply. Please let me know if you encounter any crashes thanks

j-watkiss, 2021-02-01:
I will try on tx16s when I get chance

Cheers