moschotto/OpenTX_GPS_Telemetry #9: Request for Imperial miles vs. km for distance

roligmue, 2021-01-28:
Just installed into a Frsky Horus X10. Nice work.

Looks to be working ok, but will be using in the field in the next couple of days.

Is it possible to request an option setting for distance to be displayed on the screen in Miles or Km.

Regards

moschotto, 2021-01-29:
if you just want miles, take any text editor, open the main.lua and replace these lines:

old:
wgt.gpsDtH = string.format("%.2f",wgt.gpsDtH)		
**new:**
wgt.gpsDtH = string.format("%.2f",(wgt.gpsDtH/1.609))		

old:
wgt.gpsTotalDist = string.format("%.2f",wgt.gpsTotalDist)	
**new:**
wgt.gpsTotalDist = string.format("%.2f",(wgt.gpsTotalDist/1.609))	

when you're done, copy the main.lua to the radio again and setup your widget

let know if this works

roligmue, 2021-01-29:
Thank you for taking time to reply and offer a change. I’ll give it a try and let you know.

Rick O

roligmue, 2021-01-29:
The other small change needed is to change the 2 lines that define the km labels for the display. I edited those to be mi.

After updating those lines and installing in the Frsky X10, I tested in the house and seems to be working with the proper label too. It'll be a couple of days for me to further verify and test outside.

Thank you

roligmue, 2021-01-29:
I made the changes, and installed in the Frsky X10. 
Tested in the house, and seems to be working. I'll get a test flight done in a couple days out.

Thank you

closed 2021-01-29 by moschotto