moschotto/OpenTX_GPS_Telemetry #10: Tango 2 possible?

TheSpirit, 2021-01-30:


moschotto, 2021-01-30:
Tango isn’t officially supported by OpenTX  (and  opentx companion doesn‘t work).  I can’t use the companion simulator to adapt the scripts and I don’t own a tango...

TheSpirit, 2021-01-30:
I just tried the X9 lite version and it works. 
but it doesn’t look nice since the resolution of the tango 2 is 128x96. So everything is on Top and on the bottom there is some empty space.

moschotto, 2021-01-30:
Can you share picture please ?

TheSpirit, 2021-01-30:
Sure

[image]

moschotto, 2021-01-30:
rename from .txt to .lua - as mentioned i don't own a tango... let me know how it goes and update a screen. 

BTW did you try if the scipt works in general?

[tango.txt](https://github.com/moschotto/OpenTX_GPS_Telemetry/files/5898319/tango.txt)

TheSpirit, 2021-01-30:
Looks good. 
[image]
Will send a new screenshot tomorrow with real data.

TheSpirit, 2021-01-31:
Everything seems fine. Just tested it outdoors. But the viewer seems not to work properly. After looking into the GPSpositions.txt I could see that not every line has a number and a timestamp. 
the second thing is, I am not sure what every line is showing me in the telemetry screen. Could you explain it to me in detail? With a picture?
Would be great. 
Thanks

moschotto, 2021-01-31:
Please upload a screenshot of both an mask your coordinates. Then I can take a look and upload the script to the repository. Thanks

TheSpirit, 2021-01-31:
Everything looks fine now. 

[image]

The viewer also works fine now (was my fault with messed up txt file). 
could you adopt the viewer script to the bigger size of the tango 2 screen? Maybe an empty line between the gps position lines. 

[image]

moschotto, 2021-01-31:
here you go - see attached and rename. Please provide updated screenshots if possible. 

Thanks

[GPSviewerT.lua.txt](https://github.com/moschotto/OpenTX_GPS_Telemetry/files/5900128/GPSviewerT.lua.txt)
[GPSxT.lua.txt](https://github.com/moschotto/OpenTX_GPS_Telemetry/files/5900129/GPSxT.lua.txt)

TheSpirit, 2021-02-01:
looks good

[image]

moschotto, 2021-02-01:
Thanks - can you please take a pic of the Controller with the script Running / the viewer AND the regular script an. If possible don’t mask everything- I.e.  just the first 3 values ?

TheSpirit, 2021-02-01:
i do not get exactly which 3 pictures.
What is the "regular" script?
You have only 2 scripts.

moschotto, 2021-02-01:
Pic of viewer running 
Pic of telemetry running 

I need some pics for the GitHub start page. Something like

[image]

closed 2021-02-02 by moschotto

txbone66, 2021-05-18:
They suggested that I reach out to you.  Can you tell me how you got this working on your Tango 2?  
I have the following in my Telemetry path:
[image]

but when I create a page to run that script, I get an error.  They are trying to help me with this on #16

[image]

txbone66, 2021-05-18:
Just FYI, I downloaded this Tango.txt, looked inside, moved my GPSPositions.txt to the Logs folder and then changed it to Lua and it works!@!

kirilldmt, 2022-05-28:
i updated my tango, and now i cant open script - its says Script syntax error TELEMETRY/GPSxT.lua:58 attempt to perform arith....
When drone not turned on, script opening ok