moschotto/OpenTX_GPS_Telemetry #11: Freezes on T16

jeffhendrix, 2021-02-04:
Using the version uploaded Feb 2.
Replaced the entire folder on T16. 
Add to the telemetry screen, can move in and out of the screen without any trouble until drone with GPS is powered up.
Screen stops responding, and unable to power off. Had to pull battery. Powered back up in Emergency Mode. Powered off and back on to recover radio. If drone still powered on, screen will freeze immediately. (repeat battery pull, and 2 power cycles)
Control link is through a crossfire module.

moschotto, 2021-02-04:
Can you please provide the following information?

- current installed SD card and OpenTX version
- Screenshot or pic of the discovered telemetry sensors (the script uses: GPS / Tmp2 or Sats)
-> maybe rediscovering the sesnors helps?
- how many screens did you setup? Simulator works with at least 3 screens
- did you remove the widget and THEN setup it from scatch or did you just copy the files to the SD?

Did you check this video?

https://www.youtube.com/watch?v=mqYcgPZe1KE
-> see comments:

**Mosch** 2 days ago
BTW:
does anyone facing issues with a T16 "emergency mode" after setup or turning off/on the radio? I don't own a T16 so i can't test it....

-> **TMac** 2 days ago
-> Great stuff Mosch, thanks! I've had no "Emergency Mode" issues to date using it with my TX16S

Thanks

jeffhendrix, 2021-02-06:
OpenTX Vers: 2.3.11. OPTS - crossfire, ghost, afhds3, internalmulti,
multimodule, lua, luac
SD card ver: 2.3V0035
I had 4 screens (it was the 4th). I then removed it and another screen and
set it up as the 3rd screen. Still having the same problem.
Yes, removed and readded the screen.
Followed TMac's video exactly.

Doesn't seem to freeze until it starts getting data from the quad

Tried to attach pictures, but post was rejected because it was too large.
Here is a list of the sensors
RxBt
Curr
Capa
Bat_
1RSS
2RSS
RQly
RSNR
RFMD
TPWR
TRSS
TQly
TSNR
FM
GPS
GSpd
Hdg
Alt
Sats
Ptch
Roll
Yaw

moschotto, 2021-02-06:
Strange and hard to say what causes this. I read about this issue - maybe this is similar to your issue (yaapu script):

Are you running a LUA script? There's an issue in the current firmware/nightlies which leads to emergency mode if text is drawn at the very bottom of the screen. I see this with Yaapu.

https://www.rcgroups.com/forums/showthread.php?3493859-RadioMaster-TX16S-16Ch-2-4ghz-Multi-Protocol-OpenTX-Radio-System/page284

Maybe it’s worth a try to comment the last lines in my script and try again:

	--current and last location
	
lcd.drawBitmap(drone_img, wgt.zone.x, wgt.zone.y + 150, 40)
	
lcd.drawText(wgt.zone.x + 50,wgt.zone.y + 158, wgt.coordinates_prev,LEFT + SMLSIZE + TEXT_COLOR)
	
lcd.drawText(wgt.zone.x + 50,wgt.zone.y + 180, wgt.coordinates_current,LEFT + SMLSIZE + TEXT_COLOR)
	

Sorry no more ideas currently

jeffhendrix, 2021-02-06:
Tried commenting those lines out and it still hung. Then tried commenting
out all the drawing routines, and it still hung.
I don't know lua, but I do know programming, so I started commenting out
other sections of code and I was able to narrow it down to the lines of
code that would cause it to hang.
If I comment out these 4 lines
  wgt.gpsLAT = rnd(wgt.gpsLatLon["lat"],6)
  wgt.gpsLON = rnd(wgt.gpsLatLon["lon"],6)
  wgt.gpsLAT_H = rnd(wgt.gpsLatLon["pilot-lat"],6)
  wgt.gpsLON_H = rnd(wgt.gpsLatLon["pilot-lon"],6)
it doesn't hang.

moschotto, 2021-02-06:
just and idea - maybe it makes sense to if the GPS string is longer than 5 chars:

		if(string.len(wgt.gpsLatLon["lat"]) > 5) then
		
			wgt.gpsLAT = rnd(wgt.gpsLatLon["lat"],6)
			wgt.gpsLON = rnd(wgt.gpsLatLon["lon"],6)		
				
			
			--set home postion only if more than 5 sats available
			if (tonumber(wgt.gpsSATS) > 5) then
				wgt.gpsLAT_H = rnd(wgt.gpsLatLon["pilot-lat"],6)
				wgt.gpsLON_H = rnd(wgt.gpsLatLon["pilot-lon"],6)	
			end

			wgt.update = true	
		end

jeffhendrix, 2021-02-07:
Did lots of experiments, (and still doing them)
If I remove all the calls to rnd(), just assign the variable to the
contents, and if I remove the calls to calc_Distance (just assign a hard
coded number), the widget doesn't crash.
If I restore calls to rnd() or calc_Distance (even with the rnd removed),
it will hang.
There must be something in the math routines that's causing it to hang.
Still digging.

jeffhendrix, 2021-02-10:
My crashes are coming from the rnd() and calc_Distance() functions.
If I replace this line in rnd()
return math.floor((v*10^d)+0.5)/(10^d)  with  return math.floor(v+0.5), then rnd stops crashing. I also tried replacing rnd with the sample round function in the opentx lua doc, and it crashes.
In calc_Distance(), I returned early from the function with a hard coded value. I then started moving it down in the function one line at a time and when I enable the line 
local a = math.pow(math.sin(d_lat/2.0), 2) + math.cos(LatHome*d2r) * math.cos(LatPos*d2r) * math.pow(math.sin(d_lon/2.0), 2)
the function crashes.
My best guess is maybe my radio is running out of memory or it's overflowing a variable?

jeffhendrix, 2021-02-10:
OK, so I (kind of) figured it out.
I reflashed my radio and erased the SD card. I copied the contents of a new SD card image and only installed the GPS telemetry widget. I also restored my models, and that was it.
I went out to test it and when it acquired satellites, it was working as expected (no crash)
So I'm going to slowly add back in other widgets, etc and see what happens.
So there was either too much going on on the radio or I had something not compatible or corrupted on my SD card.
Closing issue, and thanks for your work

closed 2021-02-10 by jeffhendrix

moschotto, 2021-02-10:
Hi - thanks for the debugging. What happens if you just run the GPS widget and remove the other widgets? There were MEM problems in some opentx versions...

you are using an opentx version with lots of features enabled:

OpenTX Vers: 2.3.11. OPTS - crossfire, ghost, afhds3, internalmulti,
multimodule, lua, luac

-> maybe using a less „feature rich“ FW version helps as well

jeffhendrix, 2021-02-10:
I closed the issue. I did a complete wipe and new SD card image and started
with just the GPS Telemetry and it worked.
I slowly brought back all the scripts that I like (went and re-downloaded
all of them to have the latest) and it's still working.
There were a few that I didn't install (that I wasn't using), so it was
either something outdated, corrupted, or too many.

moschotto, 2021-02-10:
Good to hear that it works for you finally... best issues are self closing issues ;)

slopingoff, 2021-02-21:
I have formatted and reloaded sd card .Installed this script on my Horus X10s and one other lua script.
Again when scrolling through to select the GPS widget for half screen the radio enters emergency mode.
I think the image area of the widget may go beyond the screen area causing emergency mode a similar issue occured with Yaapus telemetry script I seem to recall a while ago (yaapu telemetry not installed this time on my sd card for my test purposes)
Since I am not a programmer I am unable to test further and have removed this script for the time being.