moschotto/OpenTX_GPS_Telemetry #13: Emergency mode Horus X10

slopingoff, 2021-02-20:
Installed the lua script okay however when Setting up "1/2" widget screen and scrolling through to GPS widget, if I scroll past other lua widgets (i.e yapuu telemetry or 'Tadango'  graph widget) the transmitter enters emergency mode.

moschotto, 2021-02-20:
Memory issue of your radio FW you are using. Check this thread https://github.com/moschotto/OpenTX_GPS_Telemetry/issues/11

moschotto, 2021-02-26:
Works for other users like this one: https://github.com/moschotto/OpenTX_GPS_Telemetry/issues/9

closed 2021-02-26 by moschotto

slopingoff, 2021-02-27:
> 
> 
> Works for other users like this one: #9

Unfortunately It still does not explain why my installation causes Emercency  mode  only when installing this widget .I have installed Yaapus mapping widget lua (which is the only addition to my previous installation)and that does not cause emercency mode when selecting that or any other widget.
It is a shame because I like your widget as it is very well laid out lua but I cannot risk causing this emergency mode surely if the widget causes an eror it should shut down and not cause an error on the whole radio?

moschotto, 2021-02-27:
Sorry I can’t solve your problem and seems not to be the „out of display range“ problem because it works for other users with x10s. 

Try to do a clean setup and use a FW with only the features enabled you REALLY need. I don’t own a x10 and the simulator doesn’t throw any exceptions.

could be a SD card issue as well. Try comment the write logfile. 

			--write logfile		
			file = io.open(log_filename, "a")    									
			io.write(file, wgt.coordinates_current ..",".. time_power_on ..", "..  wgt.gpsSATS, "\r\n")		
			io.close(file)