moschotto/OpenTX_GPS_Telemetry #16: Error on Tango 2

txbone66, 2021-05-18:
I'm on FreedomTX 1.25.  I've added the GPSxT.lua and the BMP folder (with its contents) to the scripts/telemetry path and put GPSviewerT.lua in the scripts/tools folder.  I've added the script to a page, but when I go to that page, it says Script Syntax error TELEMETRY/GPSxT.lua:7 unexpected symbol near...  but then I can't read any more...  any suggestion?

moschotto, 2021-05-18:
Which line does throw an error? Line 7 is actually a comment ?!

txbone66, 2021-05-18:
The error on the screen is not complete.  Hoping these images can come across.  One just shows my folder where the script is located, the one shows the error message when I scroll to the page where I’m calling the script, and the setup screen pointing to the script.

moschotto, 2021-05-18:
Did you upload an pictures/ screens? I don’t own a tango 2 but this user / thread reported that it works... maybe you want to ask him for his setup?

https://github.com/moschotto/OpenTX_GPS_Telemetry/issues/10

txbone66, 2021-05-18:
[image]
[image]
[image]
There are the three pics I was trying to attach.

txbone66, 2021-05-18:
/Logs does exist, but there is only a readme.txt in that folder.

moschotto, 2021-05-18:
Open the GPSxT.lua in a text editor an remove the whole comment section, save it and try again - let’s see which line throws an error

BTW the gpspositions.txt should be in the LOGS folder respectively will be stored in the logs folder.

txbone66, 2021-05-18:
FYI must have been that logs file...

closed 2021-05-18 by txbone66