moschotto/OpenTX_GPS_Telemetry #19: Syntax error in GPSviewerX9L.lua

febbyeka, 2021-11-27:
Handset Jumper T12
TX Happymodel ExpressLRS 2.4
opentx 2.3.14

Downloaded and installed X7_X9lite version
Telemetry LUA is working well
GPSviewerX9L said "no GPSPositions.txt..." if there's no log file
After few flight, GPSPosistions.txt created. But there was syntax error when GPSviewerX9L LUA opened.
Attached GPSPositions.txt
[GPSpositions.txt](https://github.com/moschotto/OpenTX_GPS_Telemetry/files/7611879/GPSpositions.txt)

febbyeka, 2021-11-28:
After some more flies, for some reason GPSViewer is working, there is no syntax error. The coordinate is showed, but I can't scroll up or down. So basically just see four lines.
Attached gpspositions.txt
[GPSpositions.txt](https://github.com/moschotto/OpenTX_GPS_Telemetry/files/7613087/GPSpositions.txt)

moschotto, 2022-02-11:
works for me wtih the T16 simulator

[image]

[image]

closed 2022-02-11 by moschotto