moschotto/OpenTX_GPS_Telemetry #21: Request for Imperial miles vs. km for distance

roligmue, 2022-02-07:
Hi,

You gave a couple edits to get this working a while ago; display miles vs. kilometers. I had it working until i installed updates with the newer releases of the GPS Telemetry. I've since been just leaving it at kilometers which works fine.

My last attempt I was able to get Distance from home to display a Miles distance. It's the Total distance that stays at 0 in the widget. It all works fine otherwise for me otherwise.

Would you be able to share something that could get Miles displaying again?

FrSky X10s
OpenTx 2.3.14

Thank you

moschotto, 2022-02-07:
Hi, you can try to modify the calc_distance function. Convert the km to mi:

local dist = ( (6371000 * c) / 1000) / **1.609344**

https://www.rapidtables.com/convert/length/km-to-mile.html

let me know if this works for you

moschotto, 2022-02-09:
If I am testing it with the simulator it works fine,- means if moving away from home in straight line the dist home equal the total distance. Could be an issue with the GPS or telemetry update rate of opentx…

how much is it off? can you post the log file please?

roligmue, 2022-02-09:
Hi,

I did a few tests. Kind of different distances it seems, but here's my last results attached.

FYI, I turned off all telemetry logging on opentx. I was logging the telemetry metrics. I also deleted all the display pages on the X10s radio too, so now the only display screen on the radio is now the GPS LUA.

Test setup steps:
- drone power off and X10s radio off
- turned X10s radio on
- drone power on, power provided by USB port to the flight controller
- waited for satellite lock, etc.
- drove straight line then turned off drone
- turned off X10s radio
[image]
![IMG_8206](https://user-images.githubusercontent.com/78124234/153308511-db51047e-7e88-4de1
[GPSpositions.txt](https://github.com/moschotto/OpenTX_GPS_Telemetry/files/8036844/GPSpositions.txt)
-bb8a-1367380ce43f.jpg)
o

roligmue, 2022-02-09:
Here's the GPS Screen at the end of my test drive

[image]

roligmue, 2022-02-09:
[GPSpositions.txt](https://github.com/moschotto/OpenTX_GPS_Telemetry/files/8036868/GPSpositions.txt)

roligmue, 2022-02-10:
Hi,
Looking at some things in OpenTx as your timing example of updates got me thinking.

I see some update indications on in my OpenTx that may be contributing to the Total Distance calculations being off; kind of like you described.

Take a look at these short videos from my X10s.
I'm using Crossfire and my X10s has a circuit board mod installed that allows a higher baud rate that was to be able to run at the full speed of crossfire and its telemetry updates.
https://blog.seidel-philipp.de/fixed-inverter-mod-for-tbs-crossfire-and-frsky-qx7/

Anyway, maybe this helps you some by sharing these shorts videos. Note how the timing of the quad location updates are not really at 1 sec. intervals on the GPS LUA. Some are, some are not. Some are are pretty fast times.

I always thought that asterisks on the telemetry page was related to refreshing the data.  

For these, I was just running the quad plugged into USB port on table in the house.

https://user-images.githubusercontent.com/78124234/153318561-5a84ca02-7ac2-45fb-94cc-257b1f5584ca.MOV

https://user-images.githubusercontent.com/78124234/153318597-75f0bb1f-6a60-470e-a5ed-4f2112f735be.MOV

moschotto, 2022-02-10:
if you have an arduino you can simulate the same path/thing as I did. Simply connect GND and TX of the arduino to the flight ctrl GND and RX and replay the coordinates… I can provide the ino file if you want.

if this works for you, you know that it is not the radio nor the telemetry

roligmue, 2022-02-11:
Hi,

Some more testing attached with interesting results.

Did some thinking over night and a little internet browsing gave me something to try. I don't have at arduino at this point.

Summary:
I think the source of the error calculation is related to the Crossfire RF Profile link in use at the time, their updating of the telemetry metrics, plus their telemetry metric groupings by priority on update timings.

This one in particular was interesting to drive what I tested this morning. 
Kinda, validates some what I'm experiencing.
https://discuss.ardupilot.org/t/passthrough-telemetry-over-crsf-crossfire/62668

My test setup was the same, driving a straight line, doing this 2x with the same results.
1) If I set Crossfire to RF profile 50Hz, the updates of the non-critical telemetry metrics, like the GPS ones, as viewed through the OpenTx telemetry page are like a constant 1 sec heartbeats as watching the asteriks.
The Total Distance then matches the Distance from home - as I drive, and at the end.

2) If I set the Crossfire RF profile to Dynamic or 150Hz, the updates of the non-critical telemetry metrics, like the GPS, as viewed through the OpenTx telemetry page, are kind of all over the place as watching the asteriks, like I shared in a previous comment. 
The Total Distance now does not match the Distance from home - as I drive, nor at the end. And is always a lot lower.

I knew flying in RF Profile Dynamic affected some things, but thought it was more related to flight control latency.

See pics and video attached:
FullSizeRender.MOV
=======
[image]
[image]
====
[image]
[image]
====
[image]
[image]
====
[image]
[image]

moschotto, 2022-02-11:
Ok thanks for the detailed summary. So basically not related to script but rather to the crossfire update frequency… I’m closing the issue now. Checking my setup / update frequency later and let u know

closed 2022-02-11 by moschotto

roligmue, 2022-02-11:
Sounds good.

If Total Distance accuracy is important, for now I can set Crossfire RX Profile to 50Hz.

moschotto, 2022-02-11:
just FYI - works for me with both rate profiles:

crsf dynamic update:

https://user-images.githubusercontent.com/10543292/153657616-a0c6996c-740a-4866-9b32-35e32f2ea520.mp4

<br/>

crsf 150Hz update

https://user-images.githubusercontent.com/10543292/153657632-fcde418e-e094-4d25-aea6-facbe27fcb91.mp4