moschotto/OpenTX_GPS_Telemetry #22: Logfile not being written

Mike-HRC, 2022-02-15:
Hi Mosch, as I mentioned on YouTube, I am trying to get your GPS lua and GPS Stats viewer working so that I can find my planes if they get lost  - I have downloaded the scripts onto my RadioMaster Tx16S running 2.3.14 OpenTx.  The GPS lua draws the screen and after a while says "No GPS fix" - it is getting GPS data, but does not start writing to the log (even though the GPS sensor data gets a fix).

I have got a /LOGS/ folder set-up and even created a dummy GPSpositions.txt file - but this does not get written to.

I have used this GPS sensor (BN-220) before (for a logger) and it works fine, I have tried it on a very simple lua that just writes to the screen and that was working as well, so I believe that the sensor is working properly.

Please let me know any other info that will help.

Thanks for your help with this,
Mike

moschotto, 2022-02-15:
Thanks - just tried to reproduce but I can’t. Can you please make sure that all sensors have been discovered under telemetry in the radio or even delete it an rediscover?

Especially GPS, speed and alt!

logfile will be generated only if there’a a valid gps signal…

thanks

moschotto, 2022-02-15:
Sorry I forgot to add SATS… how many satellites do you see

Mike-HRC, 2022-02-15:
I put them on the same screen for confirmation

[image]
[image]

Mike-HRC, 2022-02-15:
> Sorry I forgot to add SATS… how many satellites do you see

I'm not sure - your widget says 0, but the sensor seems to have the GPS position.

Please understand - I'm sure that it will be something stupid that I have not done or done wrong - I'm not doubting your work :)

moschotto, 2022-02-15:
Sorry I meant when you check them in the telemetry page because logging etc starts:

if (tonumber(wgt.gpsSATS) >= 5) then

Mike-HRC, 2022-02-15:
I didn't know that it was shown there - I'll just check

Mike-HRC, 2022-02-15:
That must be it - I'm using a Arduino based sensor that can take inputs from things like pack voltage, current, temperatue, motor speed etc. etc. and feed them through into S.port format for FrSky receivers - possibly he is not passing all GPS field values through

moschotto, 2022-02-15:
yes that's it - what kind of sensor/arduino kind of thing is it?

Mike-HRC, 2022-02-15:
I use Pro Mini, but it will work with others - I've been using them for years on my helis as originally it took a telemetry feed from HobbyWing ESCs for things such as RPM, Temp, Current (on bigger ESCs) etc. as well as Phase sensors for RPM from other ESCs, then other bits and pieces got added as well as support for other ESCs and other Receivers (FlySky etc.).

This is the project https://github.com/dgatf/msrc for a few dollars of arduino, plus a £6 GPS I can get GPS into the TX16S as well as the lipo voltage etc. etc. - just not the number of sats it seems :)

moschotto, 2022-02-15:
excactly because it doesn't support SATS:

GPS:

    GPS Lat/Lon: GPS (0x800)
    GPS altitude: GAlt (0x820)
    GPS speed: GSpd (0x820)
    GPS compass: Hdg (0x840)
    GPS date/time: Date (0x850)

Mike-HRC, 2022-02-15:
Thank you so much for your help Moscho - sorry to waste your time.  I'll see if Daniel can include the number of Sats in the data stream. 

Is there anything that I should do to close this issue report or on YT for you to make obvious that it is my issue not yours?

moschotto, 2022-02-15:
no i'm closing it. good to see that's not my issue ;)

closed 2022-02-15 by moschotto

Mike-HRC, 2022-02-15:
Daniel made the changes to add a field with the number of satelites locked - and it's all good now - thanks again, really appreciate your help, and the excellent widget
[image]