moschotto/OpenTX_GPS_Telemetry #25: Jumper T-Pro reboot

Superiedy, 2022-04-22:
hello i tried to put your awesome project on my t-pro with edgetx, but it doesn't work properly when i open the tools gpx status it starts to restart the radio. how can i solve

moschotto, 2022-04-23:
Please follow the installation instructions - use the right folders etc. I don’t own a t-pro so I can test only with the silmulator. Edgetx 2.6.0 should work. Please double check the release that you are using….

Superiedy, 2022-04-23:
hi thanks for the quick reply. i checked and i have edgetx 2.7.0 can that be the problem? must it have version 2.6.0?

moschotto, 2022-04-24:
please follow the install instructions again. Most recent EdgeTX version works as expected:

[image]

closed 2022-04-24 by moschotto

Superiedy, 2022-04-24:
Grazie. Avevo notato che mi mancava il file TXT nella cartella log l'ho creato io manualmente, ma in seguito a questo mi si riavvia la radio appena riesco riprovo . È possibile che ho sbagliato qualcosa perché a me non genera il file TXT nella cartella log . Per ora grazie ti aggiorno