moschotto/OpenTX_GPS_Telemetry #27: GPS Log Max 84 Entrys

Topunzel, 2022-05-17:
Hello,

thank you for the nice script. It works very good on my Zorro, except that i have to delete the GPS-logfile manually when it reached 84 entrys. Otherwise it will stop recording my coordinates. Is this a bug?

Best regards

moschotto, 2022-05-17:
Hi, can’t be because the logfile will be cleared if there are more than 99 entries (log rotation). you can adjust it if you want but don’t have to.

		if ctr >= 99 then
			ctr = 0				
			--clear log
			file = io.open(log_filename, "w") 
				io.write(file, "Number,LAT,LON,radio_time,satellites,GPSalt,GPSspeed", "\r\n")		
			io.close(file)	
			
			--reopen log for appending data
			file = io.open(log_filename, "a")    			
		end

Topunzel, 2022-05-17:
Hello, the Problem is, there are never more than 84 entries 😅 it just stops when 84 are reached :)

moschotto, 2022-05-19:
The try to set it 50 and retry again (just for testing)

moschotto, 2022-05-19:
Works as expected - Zorro EdgeTX Simulator 

[image]

closed 2022-05-19 by moschotto