moschotto/OpenTX_GPS_Telemetry #31: QX7 and T-Lite version syntax error OTX2.3.15

ptwenty, 2022-07-26:
Hi thanks for this useful script

Trying to setup on QX7 and T-lite and I get syntax errors stating  " Syntax error GPSx9L.lua:8 unexpected symbol near '<' "

I've tried creating the GPSpositions.txt file manually and create LOGS folders (inside TOOLS and TELEMETRY folders, as well as sd card root)

When following the instructions for X7/X9lite I also have no ICON folder like the instructions say.  Where do I get the ICON folder and its contents?  I've created a folder called ICON and tried putting all the .bmp files inside but still get same syntax error.

I've read the other issues that are closed and none seemed to have the same error, and nobody else mentioned the ICON folder that I cannot see anywhere. 

Many thanks in advance

moschotto, 2022-07-26:
Hi - I corrected the „typo“ in the instructions 

copy the BMP folder to /SCRIPTS/TELEMETRY/BMP

ptwenty, 2022-07-26:
I had initially downloaded the HTML instead of the Lua by the way !!!! 
Thats responsible for the first syntax error I described at the top of the page

ptwenty, 2022-07-26:
I also tried the original GPSc.lua (which I still have working on my QX7) on the T-lite.
The same syntax error appears as with the newer GPS X9L.lua, and no GPS coordinates appear on the telemetry page.

moschotto, 2022-07-26:
Please follow the instructions closely. Do not use the *.c files. They will be “generated by the radio. discover the sensors use the GPS X9L.lua and setup a telemetry screen.

moschotto, 2022-07-26:
> Many thanks for the quick reply. If I start GPS Stats X9L.lua - it runs and has a flashing screen saying "more than 4 entries in GPS logfile needed"
> 
> If I try and execute GPS X9L.lua - I just get a box with "Syntax Error" and no other details.
> 
> I tried this on my T-lite when I had GPS lock ten minutes ago. My DJI goggles were receiving good GPS co-ordinates from the BF telemetry.

The viewer works only if at least 4 GPS positions are in the logfile. 

Don’t execute the GPS X9L.lua manually. It will throw an error. As mentioned- setup a telemetry screen

closed 2022-07-26 by moschotto

ptwenty, 2022-07-26:
Ah right, I’ll try going back in and re-discover sensors.

Not sure what files are the *.c files

Do you mean the GPSc.lua won’t work?

moschotto, 2022-07-26:
Radios with 128x64 pixel displays (e.g. x7 / x9 lite / Jumper T-lite)

/x7_x9lite/GPSx9L.lua
/x7_x9lite/GPS Stats X9L.lua

*.luac files are pre-compiled files. Just use the correct .lua (not .luac) scripts for your radio

moschotto, 2022-07-26:
https://youtu.be/3uNYTq7ZAFE