moschotto/OpenTX_GPS_Telemetry #33: Horis X12

WimSanders, 2023-02-18:
unfortunately I can't get it to work on my horus X12S. The program does not find my GPS. In Telemetry I see the GPS, ID 227.

moschotto, 2023-02-19:
Please follow the installation instructions closely. Here is a setup video that you can check as well

https://youtu.be/mqYcgPZe1KE

If you encounter an script ERROR paste in a screenshot or the error message. 

Thanks

closed 2023-02-19 by moschotto

WimSanders, 2023-02-19:
Thank you for the quick response,
I performed the setup exactly as described.
No error messages.
I used the T16 for my Horus X12S.
Open TX verse. 2.3.15.
![screen-2023-02-19-132640](https://user-images.githubu
[image]
sercontent.com/24661114/219945263-235466bf-16fb-43d5-ae9d-cd73ae1378a1.gif)
[image]
on the left in black the internal GPS of Horis
On the right the GPS in the model

With kind regards
Wim

WimSanders, 2023-02-19:
[image]

moschotto, 2023-02-19:
The problem is most likely caused by opentx or your radio setup. I can see that you have the same ID 227 for multiple sensors and therefore the script can’t distinguish between gps speed, altitude and coordinates. 

delete all sensors and rediscover them again

WimSanders, 2023-02-19:
There is indeed something strange going on.
I have removed all sensors and re-searched
The ID remains the same
But when I view the sensor data I see different IDs compared to the overview

[image]
[image]
[image]
[image]
[image]

moschotto, 2023-02-19:
Tested with the simulator - works as expected

[image]

WimSanders, 2023-02-19:
[image]

moschotto, 2023-02-19:
sorry but i don't know why it's not working on your side? are you using ibus?

check this
https://github.com/moschotto/OpenTX_GPS_Telemetry/issues/23

what you can try is to write a separate logfile - (open the main.lua file with a text editor).  

Add "write_log(wgt)" to the **refresh** function

For example
```
	-- set the colours
	newtext_color = wgt.options.TextColor
	newline_color = wgt.options.LineColor	
	
	get_data(wgt) 	
	write_log(wgt)
```

Add the lines between DEBUG to the **write_log** function

```
	local function write_log(wgt)

		now = getTime()    
		if wgt.old_time_write + wgt.log_write_wait_time < now then
		
			wgt.ctr = wgt.ctr + 1		
			time_power_on = SecondsToClock(getGlobalTimer()["session"])
			

			--[DEBUG  ######################
			--write debug logfile
			file = io.open("/LOGS/GPS_debug.txt", "a")    									
			io.write(file, "GPS ID:" .. wgt.gpsId ..", Alt ID:".. wgt.gpsaltId  ..", Speed ID:".. wgt.gpsspeedId  ..",SAT ID:".. wgt.gpssatId  ..",LAT:".. wgt.gpsLAT  ..",LON:".. wgt.gpsLON  ..",Sats no:"..  wgt.gpsSATS..", Alt m:".. wgt.gpsALT ..", Speed kmh:".. wgt.gpsSpeed, "\r\n")					
			io.close(file)			
			--[DEBUG  #####################**

```

the logfile /LOGS/GPS_debug.txt should look like this:

[image]

moschotto, 2023-02-19:
you can also try this GPS widget just for testing 
https://github.com/kristjanbjarni/opentx-widgets#gps-widget-widgetsgpsmainlua

WimSanders, 2023-02-19:
Like this?
		if wgt.old_time_write + wgt.log_write_wait_time < now then
		
			wgt.ctr = wgt.ctr + 1		
			time_power_on = SecondsToClock(getGlobalTimer()["session"])
			
			--[DEBUG  ######################
			--write debug logfile
			file = io.open("/LOGS/GPS_debug.txt", "a")    									
			io.write(file, "GPS ID:" .. wgt.gpsId ..", Alt ID:".. wgt.gpsaltId  ..", Speed ID:".. wgt.gpsspeedId  ..",SAT ID:".. wgt.gpssatId  ..",LAT:".. wgt.gpsLAT  ..",LON:".. wgt.gpsLON  ..",Sats no:"..  wgt.gpsSATS..", Alt m:".. wgt.gpsALT ..", Speed kmh:".. wgt.gpsSpeed, "\r\n")					
			io.close(file)			
			--[DEBUG  #####################**
							
			--write logfile		
			file = io.open(log_filename, "a")    									

			io.write(file, wgt.coordinates_current ..",".. time_power_on ..", "..  wgt.gpsSATS..", ".. wgt.gpsALT ..", ".. wgt.gpsSpeed, "\r\n")					
			io.close(file)			

			if wgt.ctr >= 99 then
				wgt.ctr = 0				

_________________________________________________________________
There is no /LOGS/GPS_debug.txt

WimSanders, 2023-02-21:
[#gps-widget-widgetsgpsmainlua](https://github.com/kristjanbjarni/o
[image]
pentx-widgets#gps-widget-widgetsgpsmainlua)
the 2 Lua files avgbatt and GPS work well.

i renamed the GPS to GPS1

[image]

make a setup of both

[image]

moschotto, 2023-02-21:
strange?! the other script uses the NAME instead of ID - can you try to change 

	--####################################################################
	--get Latitude, Longitude, Speed and Altitude
	--####################################################################		
       -- wgt.gpsLatLon = getValue(wgt.gpsId)  <= comment this line with two dashes "--"

	wgt.gpsLatLon = getValue("GPS") <= add this

moschotto, 2023-02-21:
furthermore do you have a sensor called "TMP2" or "Sats"? 

If not, the message "no GPS fix" is correct because the script can't read any satellites 
AND
if the satellites count is 2 or lower, no DATA will be displayed at all... no home pos, no current pos etc etc....

WimSanders, 2023-02-21:
it remains strange that one lua script reads the gps and the other script does not
I have been comparing the 2 lia scrips, in the 2 scripts you read the sensor in different ways

WimSanders, 2023-02-21:
--####################################################################
--get Latitude, Longitude, Speed and Altitude
--####################################################################		
   -- wgt.gpsLatLon = getValue(wgt.gpsId)  <= comment this line with two dashes "--"

wgt.gpsLatLon = getValue("GPS") <= add this

tomorrow i will try this

moschotto, 2023-02-21:
> it remains strange that one lua script reads the gps and the other script does not I have been comparing the 2 lia scrips, in the 2 scripts you read the sensor in different ways

=> please read all replies - here again:

furthermore do you have a sensor called "TMP2" or "Sats"? this is the sensor where the satellite count / amount comes from!

IF NOT the message "no GPS fix" is correct because the script can't read any satellites
AND
if the satellites count is 2 or lower, no DATA will be displayed at all... no home pos, no current pos etc etc....

WimSanders, 2023-02-21:
it is already late on my end, tomorrow i will try this

WimSanders, 2023-02-22:
the sensor I use is an original FR SKY GPS sensor
I've added the line you suggested, it makes no difference.

To check if the number of satellites is a problem I added the following line:

	--####################################################################
	--get number of satellites and GPS fix type
	--####################################################################	
	wgt.gpsSATS = getValue(wgt.gpssatId)
	
	if string.len(wgt.gpsSATS) > 2 then		
		-- SBUS Example 1013: -> 1= GPS fix 0=lowest accuracy 13=13 active satellites
		--[	Sats / Tmp2 : GPS lock status, accuracy, home reset trigger, and number of satellites. Number is sent as ABCD detailed below. Typical minimum 
		--[	A : 1 = GPS fix, 2 = GPS home fix, 4 = home reset (numbers are additive)
		--[	B : GPS accuracy based on HDOP (0 = lowest to 9 = highest accuracy)
		--[	C : number of satellites locked (digit C & D are the number of locked satellites)
		--[ D : number of satellites locked (if 14 satellites are locked, C = 1 & D = 4)		
		wgt.gpsSATS = string.sub (wgt.gpsSATS, 3,6)		
	else
		--CROSSFIRE stores only the active GPS satellite
		wgt.gpsSATS = string.sub (wgt.gpsSATS, 0,3)		
	end	
	-- WWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWW
	wgt.gpsSATS = "4"
	--- WWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWW
	--status message "guess"
	-- 2D Mode - A 2D (two dimensional) position fix that includes only horizontal coordinates. It requires a minimum of three visible satellites.)

Now the display shows 4 satellites but still no data.

moschotto, 2023-02-22:
please remove the quotes and try again - same position / location where you've already added it 

wgt.gpsSATS = 6      <= it's a number not a string

WimSanders, 2023-02-22:
you can also try this GPS widget just for testing 
https://github.com/kristjanbjarni/opentx-widgets#gps-widget-widgetsgpsmainlua

This GPS widget works well
is it possible to display the number of satellites in this widget GPS?

moschotto, 2023-02-22:
then use this widget and ask the guy who programmed it. sorry, but if you don't answer the questions or try what i said then please don't use my widget. i don't have a FR SKY GPS! 

Use Cleafllight or Betaflight, connect the GPS to the flight controller and it should work as expected.

without the information of the satellites (3D fix > 5 satellites) you can't be sure that coordinates that you are seeing on screen are actually correct and therefore it makes no sense to have "random" coordinates on screen.

moschotto, 2023-02-23:
it would be helpful if you tell me which telemetry protocol you are using and how everything is connected. 

anyway, if you just want coordinates without any sanity check like satellite count etc you have two options:

Option 1 - set the satellite count to a fixed value i.e. wgt.gpsSATS = 6 

Tested

[image]

moschotto, 2023-02-23:
**Option 2** - change this line

`lcl_drawText(wgt.zone.x + 50,wgt.zone.y + 180, wgt.coordinates_current,LEFT + SMLSIZE ,  newtext_color)`

to

`lcl_drawText(wgt.zone.x + 50,wgt.zone.y + 180, wgt.gpsLAT .. ", " .. wgt.gpsLON,LEFT + SMLSIZE ,  newtext_color)`

[image]

WimSanders, 2023-02-23:
I already suspected that my GPS is not reporting the number of satellites
[image]
[image]

moschotto, 2023-02-23:
Ok so it finally “kind of works” now… 

it would be helpful if you tell me which telemetry protocol you are using and how everything is connected.

WimSanders, 2023-02-23:
where can i find my telemetry protocol?
Does this description help?
Which GPS do you use?
I have sent an email to FR-Sky asking if their GPS reports the number of satellites

[GPS- Manual.pdf](https://github.com/moschotto/OpenTX_GPS_Telemetry/files/10813191/GPS-.Manual.pdf)

WimSanders, 2023-02-24:
[image]
for your information:

WimSanders, 2023-02-24:
which GPS sensor do you use that does display the number of satellites?

moschotto, 2023-02-24:
Again - it depends on the protocol you are using. Basically you can use any gps I.e. any ublox M8, M9 etc