moschotto/OpenTX_GPS_Telemetry #34: Reset of home won't stick

stevempotter, 2023-03-28:
When I reset telemetry, I see the home location momentarily go to zero km, but instantly goes back to the default bad home position that is 6316 km away from here. Why are my actual home coords being overwritten by bogus ones? My GPS has a good fix with 9 sats, and the correct aircraft position is showing at the bottom. But home always says
57,762289, -122.908029 which seems to be in BC, Canada. (I am in Ireland)

moschotto, 2023-03-28:
hi - it's probably because your pilot latitude and pilot longitude is just set once. either when you turn on the radio or and get a GPS fix. if the coordinates are bad, they're also stored,- in your case somewhere in canada.....

i can't do anything about that because pilot-lat and pilot-lon will be set automatically.

can you try this one and let me know if it works for you thank 

[main.lua.txt](https://github.com/moschotto/OpenTX_GPS_Telemetry/files/11094370/main.lua.txt)
(rename the file to main.lua and copy it to the radio)

stevempotter, 2023-03-29:
Yes, that seems to work. Thank you!!

closed 2023-03-29 by stevempotter

moschotto, 2023-03-29:
So you used the new script, right? And please let me know your radio and fw version.

stevempotter, 2023-03-30:
No internal GPS.  I didn't even know about this option -- Thanks! 
Where can I learn more about which GPS modules would work in Radiomaster Tx16s? Which UART connector on the radio does it use?

moschotto, 2023-03-30:
I don’t own a T16s but

https://youtu.be/euqpChlyouE

stevempotter, 2023-04-04:
Hi @moschotto, I have now installed an internal GPS in my TX16S. I can see it has a fix, sats, and HDOP in the Statistics debug window of EdgeTX. Now, how to get my radio's GPS coordinates to display in a widget? I don't see any VALUE that corresponds to internal GPS in the widget settings. It would be great if your Lua script were to keep showing updated HOME GPS coordinates if I am moving. I would not easily be able to reset stats while flying.

moschotto, 2023-04-04:
Hi - I am not planning to re-implement this feature because the widget is not meant to be a „MOVING plane/quad finder tool“ (the radio moves). Please try release 2.4 because this still uses the PILOT lat/lon for setting the home position. In version 2.4 version the home will be constantly updated but I can’t say if the pilot lat/lon will be updated as well. please keep in mind that I have no influence how the pilot lat/lon will be updated - It’s done automatically by edgetx

https://github.com/moschotto/OpenTX_GPS_Telemetry/releases

> I don't see any VALUE that corresponds to internal GPS in the widget settings. 

I can’t test and verify this because I am using a x9 lite radio. Furthermore the internal GPS is not supported by the edgetx simulator and therefore…….