moschotto/OpenTX_GPS_Telemetry #35: Radiomaster Boxer Install

mitchponton, 2023-04-02:
I have installed the Tango 2 lua in the scripts/tools and scripts/telemetry file path with the BMP folder

When I execute the Lua I get the following Syntax Error.

Did I forget something or is this just an issue with Edgetx or the new Radio?

[image]

moschotto, 2023-04-02:
Radiomaster Boxer is a "Display: 128*64 Monochrome LCD display" radio and supported with EdgeTX 2.8.1

https://oscarliang.com/radiomaster-boxer/

> EdgeTX Firmware Official Support

> EdgeTX 2.8.1 was released on the 18th of Feb 2023 which officially supports the Radiomaster Boxer radio. 

So you probably picked the wrong script. 

The script, at least the "GPSx9L.lua", works in EdgeTX Simulator (v2.8.1) just fine.

[image]

[image]

Please check the install instructions and let me know if it works.

Thanks

BTW: i can't test the Tango 2 because it isn't supported in OpenTX/EdgeTX companion

mitchponton, 2023-04-02:
I was running the 2.8.0 script. I went ahead and redid the scripts with the x9 scripts you posted. Seeing as its such a new and popular radio its nice to have the exact files to post.

On the github main it says to add a screen, is that possible to add a screen to the "home" on the radio where channel monitoring is shown?

Also still getting this syntax error

![IMG_20230401_214315 (1)](https://user-images.gith
[image]
ubusercontent.com/110638115/229373770-8a9991b1-a1f8-4786-a051-5b7ad522c63a.jpg)
?

moschotto, 2023-04-02:
Please update the radio - as mentioned above

mitchponton, 2023-04-03:
> Please update the radio - as mentioned above

Updated to 2.8.1 and put the x9 lua on the radio. Same syntax

moschotto, 2023-04-03:
since it runs in the simulator just fine and the error on your screen is truncated i've idea how to solve the problem.

please make sure that you follow the installation instructions closely

did you create a LOGS folder? <SDCARD root> /LOGS/
doube check if the GPSpositions.txt is in there, if not, create an empty file in that folder: /LOGS/GPSpositions.txt

mitchponton, 2023-04-03:
> since it runs in the simulator just fine and the error on your screen is truncated i've no idea how to solve the problem.
> 
> please make sure that you follow the installation instructions closely
> 
> * did you create a LOGS folder?  /LOGS/
> * doube check if the GPSpositions.txt is in there, if not, create an empty file in that folder: /LOGS/GPSpositions.txt
> * **did you setup a telemetry screen?** via "model settings" => page 12/12?
> 
> [image]
> 
> Update: you can't "execute" the script from the SDCARD!,- you need to setup a telemetry screen. otherwise you run into the issue:
> 
> [image]

/LOGS/ is there. I'll double check on the txt file.

I did enable the script on the telemetry page but I didn't do it correctly. I fixed that and it works now! This is absolutely the best.
[image]

closed 2023-04-03 by mitchponton

moschotto, 2023-04-03:
Ok good to know but why does it look so „broken/scrambled“? The GPSx9L.lua should fit perfectly on screen… see my screenshots in the 2nd post.

mitchponton, 2023-04-03:
> Ok good to know but why does it look so „broken/scrambled“? The GPSx9L.lua should fit perfectly on screen… see my screenshots in the 2nd post.
> 
> You are still using the wrong lua! Pay attention to script name…

I had removed everything and selected the x9 instead of x9l

Its perfect now and not scrambled.