moschotto/OpenTX_GPS_Telemetry #36: Feature request: automatically set home point upon first arm?

bede, 2023-04-27:
Hi there,
This is a really useful script that solves a problem for me, thank you! I wondered if it might be 1) useful 2) possible to automatically set the home location for the first time upon arming? After this it should require manual reset of course.

Thanks!

moschotto, 2023-04-30:
Hi - from my point of view it doesn’t make sense to set home on arming (a switch) because in case of a crash you might rearm several times (turtle mode etc)… 

„set home point once“ (initial arm) is basically implemented when you set it manually. The idea is, that you „kind of verify“ the GPS fix by looking at the coordinates and satellites…

bede, 2023-04-30:
Thanks for your reply. This is why I suggested only resetting for the first arm, I.e. when a home point has not already been set.

Agree that requiring manual verification is very sensible. I may fork your script to include e.g. altitude to facilitate flying in the event of video loss.

Thanks again for a really useful project, feel free to close.

moschotto, 2023-04-30:
Altitude is already included - see screenshots

closed 2023-04-30 by moschotto

bede, 2023-04-30:
Yes, but not at the same time as home distance – for flying blind both are
valuable at the same time

moschotto, 2023-04-30:
Every logging line has ALT included and therefore you should have it… type the coordinates in your phone and get the ALT from the logviewer