moschotto/OpenTX_GPS_Telemetry #37: Radiomaster Zorro 2.7.0 "no GPS data", BetaFlight is OK

MayanMN, 2023-07-14:
Hello, I have an issues
config:
Radiomaster Zorro, EdgeTX 2.7.0 
Lua Script GPSx9L
Log file created
Betaflight 4.4
=================
In betflight configurator I can see all data and it's fine, but there is nothing on Zorro's display or log file
page 11/12 there is only 10 sensors ( 1RSS, 2RSS, RQly, RSNR, ANT, RFMD, TPWR, TRSS, TQly, TSNR)

[image]
[image]
[image]
[image]

moschotto, 2023-07-14:
Hi,

Make sure GPS, speed, alt sensors have been discovered within the telemetry menu/screen of opentx/edgetx

So if you don’t see the mentioned sensors, the betaflight telemetry is not configured correctly. Check YouTube or the betaflight wiki how to do it. After you see the sensors mentioned above, try the script again

MayanMN, 2023-07-15:
Oh my God! 
I checked all ports, speeds, packets and other parameters. I flashed the receiver and transmitter 2 times.
It turned out you just need to enable "Telemetry output" in betaflight.
Whoever encounters a similar problem - start repairs with simple things)
All work great, thanks for your script!

closed 2023-07-15 by MayanMN