moschotto/OpenTX_GPS_Telemetry #38: GPS

sidi996, 2023-07-21:
Hello,
Frsky X10S Opentx 2.3.15. oXs Sensor with Gps.
Frsky telemetrie Sport  Sats indicates sats 113, script only the last 3. What can i do?
Dieter

moschotto, 2023-07-21:
Nochmal bitte

sidi996, 2023-07-21:
Mein GPS Sensor liefert in der Telemetrie z.B. Sats 113, im Widget wird nur die letzte Stelle angezeigt.

moschotto, 2023-07-21:
`
	
	if string.len(wgt.gpsSATS) > 2 then		
		-- smartport Example 1014: -> 1= GPS fix | 0=lowest accuracy | 14=14 active satellites
		--[	Sats / Tmp2 : GPS lock status, accuracy, home reset trigger, and number of satellites. 
		--[ 	Number is sent as ABCD detailed below. ABCD = 1014
		--[	A : 1 = GPS fix, 2 = GPS home fix, 4 = home reset (numbers are additive)
		--[	B : GPS accuracy based on HDOP (0 = lowest to 9 = highest accuracy)
		--[	C : number of satellites locked (first digit)
		--[ 	D : number of satellites locked (second digit)
		--[ 	for example: "1014"  C = 1 & D = 4 => 14 satellites
		wgt.gpsSATS = string.sub (wgt.gpsSATS, 3,6)		
	else
		--CROSSFIRE stores only the active GPS satellite
		wgt.gpsSATS = string.sub (wgt.gpsSATS, 0,3)		
	end	`

wahrscheinlich weil einfach abgeschnitten wird. welches telemetrie protokoll und welche SW (z.b. betaflight) wird benutzt?

sidi996, 2023-07-21:
Hallo, 
das  hatte ich angeschaut, komme aber damit nicht zurecht. Muß ich beim Gps suchen, weil der nur drei Stellen liefert.
X10S, opentx 2.3.15, Sport. Kein FC.

Von: moschotto
Gesendet: Freitag, 21. Juli 2023 14:50
An: moschotto/OpenTX_GPS_Telemetry
Cc: sidi996; Author
Betreff: Re: [moschotto/OpenTX_GPS_Telemetry] GPS (Issue #38)

`
if string.len(wgt.gpsSATS) > 2 then		
	-- smartport Example 1014: -> 1= GPS fix | 0=lowest accuracy | 14=14 active satellites
	--[	Sats / Tmp2 : GPS lock status, accuracy, home reset trigger, and number of satellites. 
	--[ 	Number is sent as ABCD detailed below. ABCD = 1014
	--[	A : 1 = GPS fix, 2 = GPS home fix, 4 = home reset (numbers are additive)
	--[	B : GPS accuracy based on HDOP (0 = lowest to 9 = highest accuracy)
	--[	C : number of satellites locked (first digit)
	--[ 	D : number of satellites locked (second digit)
	--[ 	for example: "1014"  C = 1 & D = 4 => 14 satellites
	wgt.gpsSATS = string.sub (wgt.gpsSATS, 3,6)		
else
	--CROSSFIRE stores only the active GPS satellite
	wgt.gpsSATS = string.sub (wgt.gpsSATS, 0,3)		
end	`
wahrscheinlich weil einfach abgeschnitten wird. welches telemetrie protokoll und welche SW (z.b. betaflight) wird benutzt?

moschotto, 2023-07-21:
was bedeutet 113? 113 Satelliten können es ja nicht sein und nochmal die Frage:

welches telemetrie protokoll und welche SW (z.b. betaflight) wird benutzt?

sidi996, 2023-07-21:
Wie geschrieben frsky X10s mit opentx 2.3.15. Protokoll ist frsky Sport. Kein Betaflight oder sonst etwas.
Ich hatte auch im Frsky Forum nachgefragt, und eine Antwort bekommen.
Offset mit -100 eingeben ....
Funktioniert.
Dieter

Von: moschotto
Gesendet: Freitag, 21. Juli 2023 16:37
An: moschotto/OpenTX_GPS_Telemetry
Cc: sidi996; Author
Betreff: Re: [moschotto/OpenTX_GPS_Telemetry] GPS (Issue #38)

was bedeutet 113? 113 Satelliten können es ja nicht sein und nochmal die Frage:
welches telemetrie protokoll und welche SW (z.b. betaflight) wird benutzt?

moschotto, 2023-07-21:
Ich habe es nicht getestet - einfach mal via Texteditor folgendes austauschen:

Alt:
wgt.gpsSATS = string.sub (wgt.gpsSATS, 0,3)		

Neu:
wgt.gpsSATS =   (wgt.gpsSATS -  (wgt.gpsSATS%10))*0.1

closed 2023-07-21 by moschotto

sidi996, 2023-07-21:
Das versuche ich morgen.Danke für deine HilfeVon meinem iPad gesendet

sidi996, 2023-07-21:
Gerade noch erfahren: Auszug aus der OXS Description:FILL_TEST1_WITH_GPS_NUMBER_OF_SAT // uncomment to activate this option; note: when there is a GPS fix 3D (or higher), then number of sat is increased by 100Von meinem iPad gesendet