moschotto/OpenTX_GPS_Telemetry #40: larger view on smaller displays with long press and show last known location on startup.

arkarkark, 2023-09-17:
Here's a feature request...

I lost video signal when flying and my plane went down, luckily I was running your code on my Tx (Jumper T-Pro). Thanks!

Unfortunately the screen is so small it was very hard to read the numbers. It'd be super nice if in the viewer app I was able to long press on a row and it could take up the whole (tiny) screen and show just the GPS coordinates in a larger font so I can enter them into google maps and find my plane. I managed to get by by taking a photo of my Tx and then zooming in and using the iOS text selection in photos magic.

It'd also be nice if the first thing the app shows when it starts up is the last place it saw your plane, rather than the first line from the file.

arkarkark, 2023-09-17:
ooh, or you could show a QR code that sent you to google maps

[luaqrcode](http://speedata.github.io/luaqrcode/) can be used to generates the codes.

a URL like [http://maps.google.com/maps?z=12&t=m&q=loc:38.9419+-78.3020](http://maps.google.com/maps?z=12&t=m&q=loc:38.9419+-78.3020) is only 60 characters long and a [level 11 qr code (61x61 pixels)](https://www.qrcode.com/en/about/version.html) can show up to 468 alphanumeric characters.

arkarkark, 2023-09-18:
I've been playing around with this and I verified that qrcodes work on my t-pro display. I can send you a pull request to show you where I'm at. Do you just copy paste the code around for each of the 4 versions?

arkarkark, 2023-09-19:
I'm considering a restructure of the code that should make it way easier for people to install and use. under root of this repo there would be `sdcard` and under that `telemetry` and `tools` then you just need to rsync  the sdcard directory to your sdcard and you'd be installed. a simple bootstrap script under tools would choose the right file to use under a subdirectory based on screen size.

I was also going to run the code through [lua-formatter](https://github.com/Koihik/LuaFormatter). I think I have a config file that matches your style pretty well.

thoughts?

arkarkark, 2023-09-19:
Progress so far...
[image]

moschotto, 2023-09-19:
Hi - as far as I know most lower performance radios doesn’t support QR code processing because of the thread execution limitation. If it takes too long the script crashes

arkarkark, 2023-09-19:
> Hi - as far as I know most lower performance radios doesn’t support QR code processing because of the thread execution limitation. If it takes too long the script crashes

weird, it works for me on my t-pro and I thought that was one of the dumber radios. I'll try it on the radio later today and see if it actually works there or not.

arkarkark, 2023-09-20:
ugh, you're right, the qrencode library file runs out of memory when I try and load it on my physical radio hardware. I guess I'll just make it display the numbers non-blinking and large when you hold down on the jog wheel.

devosthomas, 2023-09-21:
Can we try it on a Radiomaster TX16? Maybe these have more internal memory etc?

arkarkark, 2023-09-21:
@devosthomas one quick way to check is to put [this file](https://raw.githubusercontent.com/speedata/luaqrcode/master/qrencode.lua) on your sdcard under `SCRIPTS/TOOLS` and try and run it and see if you get a memory error. You will likely get a different error, but if memory is o.k. then that's a good sign. I have some thoughts on how to make it work with less memory so I'm not giving up just yet.

devosthomas, 2023-09-23:
@arkarkark [image]
Throws an error.

moschotto, 2023-09-23:
even other popular edge/open tx scripts doesn’t support QR codes because of that reason.

if you split up the qrcode lib/lua in different parts/sections, it takes too long process (alteady tried that). 

Remember that it needs to process the other code, logfile handling, as well. 

So what’s your idea on this?Currently it seems to be a little bit offtopic and not necessarily related to my scripts?

stevempotter, 2023-09-23:
I think the idea of the QR code is to save you from having to read the Tx LCD and type the coordinates into your phone to find a lost model. If you could just scan them in, that would be easier.

moschotto, 2023-09-23:
As mention a few posts earlier- works without qr 

https://github.com/moschotto/OpenTX_GPS_Telemetry/issues/28

arkarkark, 2023-09-23:
Scanning the text on the phone fails on my iPhone and t-pro. It’s especially hard since the number you want to read/scan is blinking!

I’m happy to stop discussion on this issue if you want to close it. If I get anything working I’ll send you a pr and you can decide if you want it or not. 

I think the QR code library is failing to load because of all the matrices of static data stored in them. My plan is to offload them to files that are read only when needed and then the memory cleared out. 

This would only be for the log viewing tool, not the telemetry page.

moschotto, 2023-09-23:
Sure, let me know if you have something. Greetings

closed 2023-09-23 by moschotto

moschotto, 2023-09-26:
> Here's a feature request...
>
> I lost video signal when flying and my plane went down, luckily I was running your code on my Tx (Jumper T-Pro). Thanks!
>
> Unfortunately the screen is so small it was [...]

Hi - back to the topic because I double checked it yesterday:

Larger font on 128x64 displays:
Already tried that but the coordinates won’t fit if MIDsize font will be used. Of course your can shorten the coordinates but that was not my intention 

Log viewer / last position:
already implemented but because it it a rolling log (100 lines), it could be that old coordinates are listed. To check the last position, you need to check the log file time for each coordinate (upper right in the log file viewer).

one solution could be to clear the log upon reset but in case of a rolling log you run into the same behavior…