moschotto/OpenTX_GPS_Telemetry #45: Betafpv radio lite 3 Pro syntax error

faabbiieenn, 2024-03-23:
I'm trying to set up gps widget Lua script on my Betafpv radio lite pro 3 (128x64)
[image]
[image]
Please forgive my poor English I'm french😅

moschotto, 2024-03-24:
Hi, I guess you missed step 4 of the installation instructions

faabbiieenn, 2024-03-24:
Hy thanks, but it seems that the 9lscript erase all folder as soon as the fix is done, after power off and power on all folder are there!
I use the tango scrip with modified screen (your 9xlite screen) and it works gooood!!

closed 2024-03-24 by moschotto

faabbiieenn, 2024-03-26:
[image]
[image]