moschotto/OpenTX_GPS_Telemetry #46: Where do i create the logfile

Chillinfpv, 2024-04-16:
I downloaded and installed the lua script. I discovered the new sensors and they show up in my telemetry screen. But when I try to open the gps stats lua script is says logfiles doesn't exist. Where do I create and put the log files so I can access the lua script. Also I don't have the screen with all the information on it. Idk if that will show up when I access the Lua script or what?

Chillinfpv, 2024-04-16:
[image]

Chillinfpv, 2024-04-16:
[image]

So I got it reading the satellites count but when I long press enter then back out it doesn't set the home position.

moschotto, 2024-04-16:
There is only one **big red exclamation mark** in the install instructions. Please double check and try again

closed 2024-04-16 by moschotto

Chillinfpv, 2024-04-16:
So I figured out how to create the GPSpositions.txt file. And put it in the logs folder now I'm getting a syntax error

Chillinfpv, 2024-04-16:
[image]

moschotto, 2024-04-16:
As mentioned the GPSpositions.txt will be created automatically by the script (in the directory /LOGS). Just start the telemetry script and make sure it displays the coordinates. Once some coordinates have been logged you should see the file with contents in the /LOGS folder

Chillinfpv, 2024-04-16:
My problem was i didnt know how to create a txt file. I figured thay out and my sensors were messed up. I just deleted em and discovered new. So now I have all the telemetry information but still have a syntax error 
[image]
[image]

moschotto, 2024-04-16:
Again! Delete the gps position txt file

 **the GPSpositions.txt will be created automatically by the script (in the directory /LOGS).**

Chillinfpv, 2024-04-16:
Awesome thank you! I appreciate you.