moschotto/OpenTX_GPS_Telemetry #48: no data shown / unable to discover new sensors

alfonz19, 2024-06-24:
Hi,

I suspect this is not issue with this lua script, but I was suggested otherwise by someone with more experience.

No matter how I tried, I don't get gps-related sensors and (thus) nothing is shown in this screen (all zeroes etc). Latest edgetx, latest elrs both on radio and drone. I'm able to get gps lock, coordinates are shown in osd. When having gps lock (or at least few sattelites), I deleted all sensors in radio and asked for discovering new. Only elrs related ones are shown. There is wire between fc and elrs module, it's socket, not soldered, seems OK. Telemetry is ON in receiver tab in betaflight.

Everything seems to work otherwise, but no gps-related(or battery-related) data in telemetry, and no such sensors are discovered. All I got is: 1RSS, 2RSS, RQly, RSNR, ANT, RFMD, TPWR, TRSS, TQly, TSNR

I'm guessing that either the wire itself it broken, maybe there is some magic sw steps I need to take.
Can you advice, if I'm using your project wrong or is there some hw issue indeed?

moschotto, 2024-06-24:
Maybe GPS telemetry is not activated in the betaflight config tab?

https://oscarliang.com/setup-gps-rescue-mode-betaflight/

alfonz19, 2024-06-24:
Thanks for reply.

sadly no, this is enabled as well. Gps uart is also set, but on specific baud, which was shipped from manufacturer. I tried auto and autobaud, but this didn't work work me and gps was unreliable. With this settings I'm getting almost 30 sattelites, so gps should be OK.

All I can thing of is corrupted wire between FC and rx, as everything I checked is OK. If it's not that, I'm literally hunting for unknown, which is always awesome :) That's why I wrote, since someone said, that I need your scripts in order to be able to discover new sensors.

moschotto, 2024-06-24:
Telemetry must be enabled in betaflight and if you do so, (all) these sensors must be visible when discovering new sensors:

follow this guide:
https://betaflight.com/docs/development/telemetry

alfonz19, 2024-06-24:
[image]

[image]

I also verified these:
set telemetry_disabled_pitch = ON
set telemetry_disabled_roll = ON
set telemetry_disabled_heading = ON
# Disable Battery telemetry item
set telemetry_disabled_voltage = ON
set telemetry_disabled_current = ON
set telemetry_disabled_fuel = ON
# Disable GPS telemetry item
set telemetry_disabled_altitude = ON
set telemetry_disabled_lat_long = ON
set telemetry_disabled_ground_speed = ON
set telemetry_disabled_heading = ON
# Disable Flight Mode telemetry item (BF >4.2.9)
set telemetry_disabled_mode = ON 

and all was off -- sensors should be emitted. 

But they are not.

Is there a way how to confirm, that wire is not defective and data can flow between fc and rx?
Sorry for probably abusing this thread, if you feel you helped enough (and you helped a lot indeed) feel free to say it's not related to your project and I will stop asking. If you can suggest smth instead, I'd be very grateful.

moschotto, 2024-06-24:
set telemetry_disabled_mode = ON

From my understanding this means turning off the telemetry in BF. Check and measure if there is continuity between the flight ctrl and the RX and if  RX/TX is connected correctly.

Never head issues with crossfire telemetry and BF. But with some BF versions you can encounter issues like this:

https://github.com/betaflight/betaflight/issues/13049

alfonz19, 2024-06-24:
> set telemetry_disabled_mode = ON
> 
> From my understanding this means turning off the telemetry in BF. Check and measure if there is continuity between the flight ctrl and the RX and if RX/TX is connected correctly.

sorry for not being specific. I understand it same, I checked it(and just copy-pasted names from elrs webpage), it's all off. So this should be OK.

> 
> Never head issues with crossfire telemetry and BF. But with some BF versions you can encounter issues like this:
> 
> [betaflight/betaflight#13049](https://github.com/betaflight/betaflight/issues/13049)

I still have the version 4.3.X (IIRC) with which it came from IFlight, and it seems it's earlier version, than one from 'your' link. I'm absolute n00b, I never did any hw work before, so please let me ask:  by "testing continuity" you mean testing if current can go through the wire using multimeter? How much 5V? I think I can do that, but IIUC since I can fly it, I'd expect only wire tx(on fc) <--> rx(on elrs) is damaged. And because I can fly it (in the middle of the season) I'm not too keen into doing bigger actions (which in my case still means almost any action) :)

moschotto, 2024-06-24:
Sorry but I can’t help here anymore. There are tons of videos out there how to setup different receiver types and BF telemetry. 

Make sure to check the wiring, use the latest BF fw and upgrade the receiver if necessary.

closed 2024-06-24 by moschotto

alfonz19, 2024-06-25:
> Sorry but I can’t help here anymore. There are tons of videos out there how to setup different receiver types and BF telemetry.
> 
> Make sure to check the wiring, use the latest BF fw and upgrade the receiver if necessary.

I understand, thank you for your time!